autowarefoundation/vision_pilot #9: Sence-seg model -- collect data in real urban odd

liuXinGangChina, 2024-12-10:
**Sensors:**

1. 2MP HFOV - 30° * 1
2. 2MP HFOV - 60° * 1
3. 2MP HFOV - 120° * 1
p.s. all sensor are mobile grid

**Scenarioes:**

Site
1. Urban canyon
2. Highway

Weather
1. sunny
2. rainy

Traffic status
1. rush hour
2. low

Final scenario will be combination of above conditions

**Defination of done**

- [ ] 1. sensor preparation and installation
- [ ] 2. data gathering

m-zain-khawaja, 2024-12-16:
@liuXinGangChina - could you please share what cameras sensors/manufacturer will be used for the data collection, thank you.

liuXinGangChina, 2024-12-17:
> @liuXinGangChina - could you please share what cameras sensors/manufacturer will be used for the data collection, thank you.

Morning Zain

Already update the manufactor in the post above，feel free to reach me when anything is needed

Best Regards

心刚

closed 2025-03-06 by m-zain-khawaja