autowarefoundation/vision_pilot #14: PathDet Dataset Curation - CULane

m-zain-khawaja, 2024-12-13:
Parse the [CULane Dataset](https://xingangpan.github.io/projects/CULane.html) which can be downloaded from this [link](https://drive.google.com/drive/folders/1mSLgwVTiaUMAb4AVOWwlCD5JcWdrwpvu) to create a dataset comprising input images in PNG format and a single drivable path as the ground truth, derived as the mid-line between the left/right edges of the ego lane.

Please create a script called process_culane.py in this folder (https://github.com/autowarefoundation/autoware.privately-owned-vehicles/tree/main/PathDet/create_path/CULane) which is responsible for creating the ground truth and saving the data.

The ground truth drivable path information should be stored as a list keypoints defining the path in a JSON format. The drivable path keypoints should be sampled with a vertical (y image axis) displacement of 10 pixels to match the ground truth format used in other datasets.

Please ensure that keypoints are stored in relative coordinates, where the top left-most corner of the image is 0,0 and the bottom right-most corner of the image is 1,1, and all other coordinates are floating point values in the range (0,1) for x,y directions.

Please ensure ground truth images are stored in PNG format

Please also save a semantic drivable path mask which is drawn upon the input RGB image alongside the ground truth drivable area for data auditing purposes as well as a binary drivable path mask in PNG format, where drivable path pixels are assigned a value of 255, and non-drivable path pixels are assigned a value of 0.

Data Summary:
RGB image in PNG Format
Drivable path keypoints in JSON Format
Binary Drivable Path Mask in PNG format
Semantic Drivable Path Mask draw on top of RGB image in PNG format (not used during training, only for data auditing purposes)

TranHuuNhatHuy, 2024-12-16:
2024/12/16

- [ ] Crop out bottom part containing the car (should be 190 pixels height, so we have 400 px in height remaining, but please test it carefully).
- [ ] Crop width left and right so new width : new height = 2 : 1.
- [ ] Update relative coords to account for new image size.
- [ ] Temporal downsampling by a factor of 5 (basically take 1 skip 4).

- [ ] Refactor the JSON and output dir structure as in slides p. 22.
[image]

TranHuuNhatHuy, 2024-12-23:
## 2024/12/23

### Housekeeping minor stuffs

- [ ] Bottom crop : 160.
- [ ] Adjust horizontal crop to be 2:1 ratio.
- [ ] Make it default arg.
- [ ] Default sampling rate to `5`.
- [ ] Also update README.md

### Post-processing

- [ ] Data audit
- [ ] Kaggle-ing it the same way with TuSimple

m-zain-khawaja, 2024-12-27:
Completed

closed 2024-12-27 by m-zain-khawaja