autowarefoundation/vision_pilot #21: PathDet Dataset Curation - Comma2k19

m-zain-khawaja, 2024-12-17:
## Description

Parse the [Comma2k19 Dataset](https://github.com/commaai/comma2k19) to create a dataset comprising input images in PNG format and a single drivable path as the ground truth, derived from the recorded poses of the ego-vehicle projected into the image plane.

Please create a script called process_comma2k19.py in this folder (https://github.com/autowarefoundation/autoware.privately-owned-vehicles/tree/main/PathDet/create_path/Comma2k19) which is responsible for creating the ground truth and saving the data.

The ground truth drivable path information should be stored as a list keypoints defining the path in a JSON format. The drivable path keypoints should be sampled with a vertical (y image axis) displacement of 10 pixels to match the ground truth format used in other datasets.

Please ensure that keypoints are stored in relative coordinates, where the top left-most corner of the image is 0,0 and the bottom right-most corner of the image is 1,1, and all other coordinates are floating point values in the range (0,1) for x,y directions.

Please ensure ground truth images are stored in PNG format

Please also save a semantic drivable path mask which is drawn upon the input RGB image alongside the ground truth drivable area for data auditing purposes as well as a binary drivable path mask in PNG format, where drivable path pixels are assigned a value of 255, and non-drivable path pixels are assigned a value of 0.

Data Summary:
RGB image in PNG Format
Drivable path keypoints in JSON Format
Binary Drivable Path Mask in PNG format
Semantic Drivable Path Mask draw on top of RGB image in PNG format (not used during training, only for data auditing purposes)

## Note
Please work on the **[path-det-dataset-curation](https://github.com/autowarefoundation/autoware.privately-owned-vehicles/tree/path-det-dataset-curation)** branch

m-zain-khawaja, 2024-12-27:
@siddas27 - I have created a new branch for this development called '21-pathdet-dataset-curation-comma2k19' - please fork this repo to your personal github, push your latest changes to the new branch on your github and then submit the pull request once completed for review. Thanks!

siddas27, 2025-01-04:
PR: #34

m-zain-khawaja, 2025-01-22:
PR Merged - thank you @siddas27

closed 2025-01-22 by m-zain-khawaja