autowarefoundation/vision_pilot #22: LaneDet Dataset Curation - TuSimple

m-zain-khawaja, 2024-12-20:
### Description:

Parse the [TuSimple lane detection dataset](https://www.kaggle.com/datasets/manideep1108/tusimple?resource=download) to create a dataset comprising input images in PNG format and lane masks as the ground truth.

- Please create a script called **`process_tusimple.py`** in this folder (https://github.com/autowarefoundation/autoware.privately-owned-vehicles/tree/main/LaneDet/create_lane/TuSimple) which is responsible for creating the ground truth and saving the data. 
- The ground truth lane information should be stored as a list of lanes, where each lane list comprises a list of keypoints, defining the lane line - in a JSON format.
- Please also classify which lanes are the ego lanes - we need to know the ego left lane, and we need to know the ego right lane - so there should be 3 classes in total (ego left, ego right, and other)
- Please ensure that keypoints are stored in relative coordinates, where the top left-most corner of the image is 0,0 and the bottom right-most corner of the image is 1,1, and all other coordinates are floating point values in the range (0,1) for x,y directions
- Please also save a binary lane mask for all lanes (lane pixels are 255 - white, and background pixels are 0 - black)
- Please ensure ground truth images, and binary lane mask are stored in PNG format
- Please also save a lane visualization mask in which the lane lines are visualized by being drawn upon the input RGB image for data auditing purposes.

### Data Summary:

- RGB image in PNG Format
- Lane lines list with lane keypoints in JSON Format
- Binary Lane Mask (black and white)
- Lane Visualization Mask (not used during training, only for data auditing purposes)

m-zain-khawaja, 2024-12-27:
@devang-marvania - I have created a new branch for this development, it is called 'lanedet-dataset-curation-tusimple' - please fork the repo to your personal github, push your changes to this branch on your github and then submit the pull request for review. Thanks!

devang-marvania, 2024-12-27:
> <autowarefoundation/autoware.privately-owned-vehicles/issues/22/2564061722
> @github.com>

Got it, thanks Zain!

devang-marvania, 2025-01-03:
Hi Zain, 

I am ensuring my understanding is correct for expected Binary Lane Mask and Lane Visualization Mask outcomes. Let me know if attached example is correct or needs changes:
[Example.pptx](https://github.com/user-attachments/files/18303040/Example.pptx)

m-zain-khawaja, 2025-01-04:
Hi @devang-marvania - thanks for sharing the powerpoint and for your message, here is a simple way to think about it:

## Binary Lane Mask:
- This is a black and white image used for tasks such as semantic segmentation
- We want all those pixels where any lanes are present to be white (i.e. assigned a value of 255)
- All remaining pixels should be black (i.e. assigned a value of 0)

### Here is an example image of a binary lane mask
[image]

## Lane Visualization Mask:
- This is similar to what you showed in the power point, except we also want to visualize all the other lanes, not just the ego right and ego left lanes.
- You can assign one colour to the left ego lane (e.g. Green), one colour to the right ego lane (e.g. Blue), and one colour to all other lanes (e.g. Yellow)

### Here is an example of a lane visualization mask 
(Please note that in this example, each individual lane is assigned a separate colour whereas you should please follow a colouring scheme as specified in the bullet above)

[image]

devang-marvania, 2025-01-05:
> <autowarefoundation/autoware.privately-owned-vehicles/issues/22/2571308114

Sounds good, thank you!

m-zain-khawaja, 2025-01-08:
Pull request has been merged - thanks!

closed 2025-01-08 by m-zain-khawaja