autowarefoundation/vision_pilot #25: SceneSeg model -- Share test results of real urban scene data collected by different cameras

cyn-liu, 2024-12-23:
### 1. 2MP HFOV - 30° * 1 leopard-ar0233-nvp2650

[image]

### 2. 2MP HFOV - 120° * 1 leopard-ar0233-nvp2650

[image]

m-zain-khawaja, 2024-12-23:
@cyn-liu - thank you for sharing - excited to see these results on roads the network has never seen before on a country outside of the training data. Is there also a link to the videos somewhere? I also suggest that for the wide-angle camera, to crop the bottom portion of the image with the car bonnet, since during training the bonnet region was specifically cropped out.

cyn-liu, 2025-01-07:
> Is there also a link to the videos somewhere?

I have re-edited the link above, and and when you click on the corresponding image will open the video link.

m-zain-khawaja, 2025-01-07:
That's great - thanks for sharing, I really appreciate it. For the wide-angle camera 120-degree, the minor artefacts on the bonnet will go away if the bonnet pixels are cropped out before running inference

cyn-liu, 2025-01-08:
> For the wide-angle camera 120-degree, the minor artefacts on the bonnet will go away if the bonnet pixels are cropped out before running inference

Thank you for your suggestion. I will crop the wide-angle 120 degree image data and then running inference，and  results will be shared.

cyn-liu, 2025-01-08:
Replaced the original test video with the inference result of the cropped image.

m-zain-khawaja, 2025-01-08:
Thanks so much - I really appreciate it :+1:

Comparing both videos - cropped and uncropped, it appears that objects on the far left/right edges of the video are better detected in the uncropped version, however, the artefacts are present on the bonnet. In the cropped version, the bonnet artefacts disappear but sometimes the far/left objects are detected with less accuracy. 

So, the best recipe is probably -> run inference on uncropped image -> then crop bonnet in post processing

It's so great to see these real-world automotive camera/mounting data because these implementation recipes can only be learned this way :relaxed:

cyn-liu, 2025-01-08:
> So, the best recipe is probably -> run inference on uncropped image -> then crop bonnet in post processing.

Thank you for your suggestion. 

I plan to upload three comparison schemes:

1. run inference on origin uncropped image
2. run inference on origin cropped image
3. run inference on origin uncropped image and then remove the car crop bonnet through post-processing.

m-zain-khawaja, 2025-01-08:
Excellent - thank you so much!

m-zain-khawaja, 2025-01-09:
Thank you very much for sharing these comparisons - I really appreciate it

m-zain-khawaja, 2025-02-04:
@cyn-liu - is there a plan to do further data collection and testing?

closed 2025-03-06 by m-zain-khawaja

liuXinGangChina, 2025-03-10:
Hello, Khawaja san,

Latest test result for ScensSeg under foggy weather is released in the discussion channel

[image]

Please feel free to leave your comment [here](https://github.com/orgs/autowarefoundation/discussions/5858)

Have a nide day!

心刚

m-zain-khawaja, 2025-03-10:
Thank you very much for your efforts to collect and share this data, it is really appreciated.

m-zain-khawaja, 2025-08-13:
Closing, no new updates received