autowarefoundation/vision_pilot #46: EgoLanes Dataset Curation - BDD100K

m-zain-khawaja, 2025-02-04:
### Description:

Parse the [BDD100K lane detection dataset](https://github.com/bdd100k/bdd100k) to create a dataset comprising input images in PNG format and lane masks as the ground truth. [Link to academic paper](https://arxiv.org/abs/1805.04687)

- Please create a script called **`process_bdd100k.py`** in this [folder](https://github.com/autowarefoundation/autoware.privately-owned-vehicles/tree/main/EgoLanes/create_lane/BDD100K) which is responsible for creating the ground truth and saving the data. 
- Please also create a README.md file describing the dataset and script details, in the same folder reference in the point above
- The ground truth lane information should be stored as a list of lanes, where each lane list comprises a list of keypoints, defining the lane line - in a JSON format.
- Please also classify which lanes are the ego lanes - we need to know the ego left lane, and we need to know the ego right lane - so there should be 3 classes in total (ego left, ego right, and other)
- Please ensure that keypoints are stored in relative coordinates, where the top left-most corner of the image is 0,0 and the bottom right-most corner of the image is 1,1, and all other coordinates are floating point values in the range (0,1) for x,y directions
- Please also save a binary lane mask for all lanes (lane pixels are 255 - white, and background pixels are 0 - black)
- Please ensure ground truth images, and binary lane mask are stored in PNG format
- Please also save a lane visualization mask in which the lane lines are visualized by being drawn upon the input RGB image for data auditing purposes.

### Data Summary:

- RGB image in PNG Format
- Lane lines list with lane keypoints in JSON Format
- Binary Lane Mask (black and white)
- Lane Visualization Mask (not used during training, only for data auditing purposes)

### Example of same task on a different dataset:
Please review this work to see an example of dataset parsing for another dataset: https://github.com/autowarefoundation/autoware.privately-owned-vehicles/issues/40

m-zain-khawaja, 2025-02-05:
@aadarshkt - please use the **egolanes-dataset-curation---bdd100k** branch

aadarshkt, 2025-02-10:
Update - Ground truth information is saved in JSON file formate with normalized keypoints, binary masks which was already available in the dataset is saved in Egolanes/create_lane/BDD100k/output/binary_masks. I have attached images for reference. Code can be fount at [Link to git diff](https://github.com/aadarshkt/autoware.privately-owned-vehicles/commit/d4953608952ef9f4e1d5336e8c0eca0e1791a373)

[image]

[image]

Next task - To classify lanes as Ego Left Lane, Ego right lane, and other lanes.

aadarshkt, 2025-02-17:
Update - 

1. Changed the format of the JSON to match reference format.
2. First version of annotated grouth truth images with lanes classified as {blue, green, yellow} -> {right, left, other_lanes}.

**JSON**
[image]

**Annotated Images**

[image]

[image]

[image]

**Wrongly Annotated Images**

[image]

[image]

[image]

In progress - 

1. Cropping of image to match reference crop dimensions.
2. Finding reasons for wrongly annotated images.
3. Matching images names format with reference JSON format.

aadarshkt, 2025-03-03:
Update - 

1. Cropping of annotated image and keypoints in ground truth data is completed.
2. Images are not correctly annotated.
3. Images names are in serial order. 
4. Data uploaded on kaggle.(Currently uploading)

aadarshkt, 2025-03-10:
Update - 

1. Removed all the lanes with vertical lanedirection.
2. Tried to merge two adjacent right or left ego lanes but currently not successful.

aadarshkt, 2025-03-14:
Update - 

Used sorted anchor points and merge threshold, which is 5% of image width to merge lanes.

Explaination of logic -

Any lane line which has anchor point within 5% of Image width is chosen for merging. Merging is done as follows. 

Merging follows interpolation logic in which, several points along y_max and y_min is chosen in linear fashion to get corresponding x coordinated for both lanes lines, these x coordinates are then averaged to (x,y) tuples for merge lanes. 

Detailed explanation.
y = ymin + (ymax - ymin) * (t) / (req_points), where t varies between 0 to req_points, and req_points is the number of required points for interpolation.

This y is used to calculate corresponding x for each lane line using slope and y coordinate. Slope is calculated on the basis of nearest two y coordinates in the lane line.

X coordinates for both lanes lines are then averaged to get single tuple (x,y) of lane line.

Results - 

Before 

[image]

[image]

After merging lanes.

[image]

[image]

aadarshkt, 2025-03-16:
Update - 

Using threshold as angle between extended lane lines and x-axis, gives better results. 
Before
[image]

After
[image]

aadarshkt, 2025-03-24:
**Update** - 

Merged all lane lines. [Link to artifacts that contain 100 images with and without merged lane lines](https://www.kaggle.com/datasets/aadarshkumartiwari/bdd100k-merge-lane-lines-and-ground-truth-dataset)

**Edge cases** - 

Three lane lines together are not being merged.
Before - Notice three extreme right lane lines
[image]

After - 
[image]

Lanes with low angles with x-axis have high distance between anchor pionts thus they cannot be merged. For example lane with id (0,1) and (6,7) will not be merged. Current anchor distance threshold is 15% of Image width.

[image]

closed 2025-04-23 by m-zain-khawaja