autowarefoundation/vision_pilot #53: Augmentations - EgoPath Network

m-zain-khawaja, 2025-03-06:
## Description:

Please **contribute** to the existing **`augmentations.py`** file which can be found under [Models/data_utils/augmentations.py](https://github.com/autowarefoundation/autoware.privately-owned-vehicles/blob/main/Models/data_utils/augmentations.py) to add support for data augmentations related to the EgoPath network. To apply data augmentations, we rely upon the [Albumentations Library](https://albumentations.ai/docs/) - please carefully reference the documentation and example showing augmentations for keypoints which can be found at this [link](https://albumentations.ai/docs/getting_started/keypoints_augmentation/)

- Add a new dataset type to the class initializer **`Literal`** variable **`data_type`**, called **'KEYPOINTS'** 
- Update the dataset type checker in line 19 to include checking for data of type **'KEYPOINTS'**
- Create a transform called **`self.transform_shape_keypoints`** which should apply resizing of the image to width of 640px and height of 320px, horizontal flip with probability of 0.5 and a small rotation of +/- 10 degrees
- Create a transform called **`self.transform_shape_keypoints_test`** which should only apply resizing

- Create the following helper functions:
  1. **`def setDataKeypoints(self, image, ground_truth)`** - which should set the class variables self.image and self.augmented_iamge to be the input image, and the class variable self.ground_truth and self.augmented_data to be the ground_truth 
  2. **`def applyTransformDepth(self, image, ground_truth)`** - which should apply the transformation, depending on whether it is for training or validation, and return the augmented image and augmented ground truth

TranHuuNhatHuy, 2025-03-16:
@m-zain-khawaja pardon me if I'm wrong, but shouldn't it be `def applyTransformKeypoint()` instead of `def applyTransformDepth()` ?

m-zain-khawaja, 2025-03-16:
Yes, it should be `def applyTransformKeypoint()` - apologies for the typo

m-zain-khawaja, 2025-03-30:
PR merged with main - thanks @TranHuuNhatHuy

closed 2025-03-30 by m-zain-khawaja