autowarefoundation/vision_pilot #64: EgoPath Dataset Rework - Unifying structure and format for all 6 datasets

TranHuuNhatHuy, 2025-03-27:
# I. Background

EgoPath's processed data was parsed from 6 different lane-detection datasets:
- TuSimple
- CULane
- CurveLanes
- BDD100K
- Comma2k19
- ROADWork

Since they are lane detection datasets, we acquired the lane/line information, and derived the ego path based on a heuristic approach of determining the 2 left and right ego lines, then calculating the ego path based on x-wise mean of those 2 lines. Although we share mostly the same approach, due to the fact that this is a collective effort from different members, the format and structure of the processed datasets are quite different from each other, making it troublesome in downstream tasks.

Thus, after collecting ALL of the 6 processed datasets, I am refactoring them, structure-wise and format-wise, in order to make them as analogous as possible.

# II. Overview

## 1. Unified file structure

Each dataset will bear the same file structure as followed:

```
|------ processed_<dataset name in UPPERCASE>/              # i.e. processed_CURVELANES
|              |------ image/
|              |------ segmentation/
|              |------ visualization/
|              |------ drivable_path.json
```

## 2. Unified JSON format

The ego path label file, `drivable_path.json` of each dataset will now have the format as followed:

```
{
    <6-digit frame id> : {
        "drivable_path": [
            [<x_1>, <y_1>],
            [<x_2>, <y_2>],
            [<x_3>, <y_3>],
            .....................
            [<x_n>, <y_n>],
        ],
        "img_height": <h>,
        "img_width": <w>
    },
    .................................
}
```

---

Since this is a no-code refactor effort rather than pushing some lines of code onto the repo, I leave it as an issue like this rather than a PR. @m-zain-khawaja after I finish this and you have verified it, this issue can be closed.

TranHuuNhatHuy, 2025-03-28:
## 1. Comma2k19

### a. File structure

- Restructure dataset file tree, currently as in below image, to the unified structure.

[image]

### b. JSON format

- Refactor `drivable_path.json` to the unified format.
- Normalize all coords according to their corresponding frame's dimensions, as current version seems to forget to do so.
- Trim points whose `y > img_height`.
- Rename ALL of frame IDs, from `00001` (5-digit, starts from 1) to `000000` (6-digit, starts from 0). This occurs both in all 3 folders `image`, `segmentation` and `visualization`, as well as in `drivable_path.json`.

TranHuuNhatHuy, 2025-03-28:
## 2. CurveLanes

All of CurveLanes's problems have been fixed from the data parsing process `EgoPath/create_path/process_curvelanes.py`, addressed in #63 , instead of manual changes like in Comma2k19.

### a. JSON

- Get rid of all points whose `y > img_height`.
- Interpolate more points into each lines, to ensure they survive through the resize/crop operations, which will retain the highest characteristics of their shape. It works, surprisingly well actually.

### b. File structure

- File structure stays the same, but all images from those 3 folders are regenerated, which bear the above updates.

TranHuuNhatHuy, 2025-03-28:
## 3. ROADWork

### a. JSON

- Refactor `drivable_path.json` to the unified format.
- Rename ALL of frame IDs, from 00000 (5-digit) to 000000 (6-digit). This occurs both in all 3 folders image, segmentation and visualization, as well as in drivable_path.json.

### b. File structure

N/A.

TranHuuNhatHuy, 2025-03-28:
## 5. CULane

The aforementioned unified format/structure are based on CULane. Thus, there are no changes necessary.

However there used to be a frame ID mismatch issue, which has been fixed. You can refer to my PR here #60 .

TranHuuNhatHuy, 2025-03-28:
## 6. BDD100K

### a. JSON

- Rename ALL of frame IDs, from 00000 (5-digit) to 000000 (6-digit). This occurs both in all 3 folders image, segmentation and visualization, as well as in drivable_path.json.

### b. File structure

N/A.

TranHuuNhatHuy, 2025-03-28:
Meeting update with @m-zain-khawaja:

- Point order in `drivable_path.json` should be unified: bottom up, biggest y-coord first.
- Dataset folder doesn't need that `processed_` prefix.

TranHuuNhatHuy, 2025-03-28:
Quick fixes on working session:
- [ ] ROADWork
- [ ] BDD100K
- [ ] TuSimple
- [ ] Comma2k19
- [ ] CULane
- [ ] CurveLanes

m-zain-khawaja, 2025-03-30:
Kaggle upload has been completed, thanks @TranHuuNhatHuy

closed 2025-03-30 by m-zain-khawaja