autowarefoundation/vision_pilot #67: EgoPath - Manual Data Audit

m-zain-khawaja, 2025-03-31:
## Description:

### **_"A neural network is only as good as its data"_**

The purpose of this task is to perform a manual data audit of the datasets being used to train the EgoPath neural network. To begin, please download the new and improved EgoPath Master Dataset from the following [Kaggle Link](https://www.kaggle.com/datasets/210a3bb0d73ac766183e05232da1f4521d636305f0f0671e6d3eaaa2c5223d6b)

- For data auditing, please inspect the visualization folder of a given dataset and **_view each visualization image_**. 

- Confirm that the annotated drivable path matches the expectation of where you think a human driver would navigate if they had to "follow the road" as shown in the visualization image. 

- Please create a CSV file called `audit_discard.csv` and record the frame ids (e.g. 000000, 000010 etc.) of any visualization frames which you feel are erroneous in this CSV file

- In particular, please remove any frames where the drivable path is based on a lane change or the vehicle is making a turn at a junction - the focus of the dataset should be on following the road - it is okay if the road is curvy

### Assignees:

- @TranHuuNhatHuy - could you please do a data audit of the **CurveLanes dataset**
- @sarun-hub - could you please do a data audit of the **ROADWork datset**
- @auo355 - could you please do a data audit of the **CULane dataset**
- @aadarshkt - could you please do a data audit of the **BDD100K dataset**
- @pranavreddy23 - could you please do a data audit of the **Comma2k19 dataset**

auo355, 2025-04-14:
Find Attached, the CSV file for rejected images of CULANE sub-dataset of EgoPathDataSets

[audit_discard.csv](https://github.com/user-attachments/files/19743364/audit_discard.csv)

sarun-hub, 2025-04-18:
Hello team, this is a csv for rejected frames of RoadWork dataset for EgoPath Training. Check it out!
Total of 914 frames are rejected out of 4800 frames -> 19%,

[final_auto_discard.csv](https://github.com/user-attachments/files/19807667/final_auto_discard.csv)

aadarshkt, 2025-04-24:
Hello team, this CSV file contains all the reject frames for BDD100k dataset for EgoPath Training. Intially I was bit more conservative, so I will review again the rejected frames to reduce it by maybe ~10%. Currently total 999 frames are rejected out of 24.5k frames.

[audit_discard.csv](https://github.com/user-attachments/files/19887202/audit_discard.csv)

TranHuuNhatHuy, 2025-04-24:
Manual audit finished.
- Total frames: 19575
- Accepted: 16660 (85.1%)
- Rejected: 2915 (14.9%)
Attached file.

[discarded_set.csv](https://github.com/user-attachments/files/19888816/discarded_set.csv)

pranavreddy23, 2025-04-25:
Hi Team.

I used @TranHuuNhatHuy 's app and went through data auditing one more time and it made things easy to add few more frames to the list and I am attaching the final version of data audit. Please use the below csv file .Thanks for Tran for the tool.

[dataAudit.csv](https://github.com/user-attachments/files/19915110/dataAudit.csv)

auo355, 2025-05-03:
Good day all, I have audited the TUSIMPLE Folder. The JSON File is attached. 

[tusimple_accept_reject.json](https://github.com/user-attachments/files/20024178/tusimple_accept_reject.json)

Some additional files to be rejected which was not captured in the JSON file is captured in the csv file attached. 

[additional_rejected_files_for_TUSIMPLE.csv](https://github.com/user-attachments/files/20024184/additional_rejected_files_for_TUSIMPLE.csv) This is because, the python app for accept/reject does not go backward for correction. So I had to add the names of the additional files to be rejected in the csv document

m-zain-khawaja, 2025-05-08:
Thank you everyone for your efforts! 👍