autowarefoundation/vision_pilot #78: EgoPath Adversarial Test Set

TranHuuNhatHuy, 2025-04-25:
Please create a test set of internet images for EgoPath with adversarial and challenging highway data including, approx. 5 samples from each of the below categories:

- Faded Lanes
- Dots and dabs
- Snow on the road
- Rainy road conditions
- Foggy road conditions
- Night time
- Driving through tunnels
- Sun glare
- Heavy traffic
- Marks/repairs on road (black bitumen lines)
- High curvature road
- Bizarre - (strange colour of lanes, road markings)
- Construction zones (traffic cones, barriers, diversions) about 10 of them

m-zain-khawaja, 2025-05-02:
@TranHuuNhatHuy - initial adversarial test set has been added to this [google drive link](https://drive.google.com/drive/folders/1dKtcwE88tXPT0-jEFMLmsPg7xGDhotLy?usp=sharing)

m-zain-khawaja, 2025-05-08:
Initial adversarial dataset is sufficient for testing at this stage

closed 2025-05-08 by m-zain-khawaja