autowarefoundation/vision_pilot #81: Network Training - EgoPath Network

m-zain-khawaja, 2025-04-29:
# Description:

Each open-source contributor can train the EgoPath network and conduct various experiments to achieve the best results on the validation data split. Beneficial changes from each set of experiments can be incorporated by other open-source contributors for new experiment batches. 

**Please ensure you have an Nvidia GPU to conduct training experiments.**

-------------------------------------------

## Getting started:

1. To begin, please make sure that you pulled the latest code changes from this repo and have downloaded the [EgoPath dataset](https://www.kaggle.com/datasets/1c6a68c61152e94660b42c45b87296308abf1a4c0838bc4dbddda8066c6821db) and unzip it.
2. Create a folder in your repo directory (not added to the git tracking) under the file path /Models/saves/SceneSeg, then download [SceneSeg pre-trained checkpoint file](https://drive.google.com/file/d/1vCZMdtd8ZbSyHn1LCZrbNKMK7PQvJHxj/view?usp=sharing) and save it in this folder
3. Create a folder in your repo directory (not added to the git tracking) under the file path /Models/saves/EgoPath - this is where the EgoPath checkpoint files will be saved during training
4. There is a requirements.txt file in the Models/training folder which specifies the libraries which you will need to run the training script and conduct training experiments, you can install the requirements simply as per:

```bash
pip install -r requirements
```

**Please note, you may need to install Pytorch manually based on your CUDA version, you can visit this [website](https://pytorch.org/get-started/locally/) to learn more**

-------------------------------------------

## Running the main train script:

The main train script is located under /Models/training/train_ego_path.py. From your terminal, navigate to the /Models/training folder, then you can run the main train script as follows, passing in three command line parameters 

- **-r** the path to the unzipped downloaded EgoPath data
- **-b** the path to the downloaded SceneSeg *.pth file
- **-s** the folder where you want the EgoPath *.pth files to be saved during training

#### For example:

```bash
python3 train_ego_path.py -r /file/path/to/data/EgoPathDatasets/ -b /repo/Models/saves/SceneSeg/SceneSeg.pth -s /repo/Models/saves/EgoPath/
```

If you have stopped the training after having started it, and you want to resume training from that saved point, instead of passing in the **-b** parameter, pass in the **-c** parameter with the path to the saved EgoPath *.pth checkpoint file

#### For example, to continue training from a saved checkpoint:

```bash
python3 train_ego_path.py -r /file/path/to/data/EgoPathDatasets/ -c /repo/Models/saves/EgoPath/EgoPath.pth -s /repo/Models/saves/EgoPath/
```
-------------------------------------------

## Modifying parameters as part of training experiments:

There are a number of parameters which you can modify when setting up your training experiment. **You will need to edit these parameter values directly in the train_ego_path.py file before running the main training script.** You can find the modifiable parameters in the train_ego_path.py file between **lines 154 - 215**

The modifiable parameters are as follows:

### SCALE FACTORS

```python
    # ========================= SCALE FACTORS ========================= #
    # These scale factors impact the relative weight of different
    # loss terms in calculating the overall loss. A scale factor
    # value of 0.0 means that this loss is ignored, 1.0 means that
    # the loss is not scaled, and any other number applies a simple
    # scaling to increase or decrease the contribution of that specific
    # loss towards the overall loss

    ENDPOINT_LOSS_SCALE_FACTOR = 0.0          
    MIDPOINT_LOSS_SCALE_FACTOR = 0.0
    GRADIENT_LOSS_SCALE_FACTOR = 1.0
    CONTROLPOINT_LOSS_SCALE_FACTOR = 1.0
```

### GRAD_LOSS_TYPE

```python
    # ======================== GRAD_LOSS_TYPE ======================== #
    # There are two types of gradients loss, and either can be selected.
    # One option is 'NUMERICAL which calculates the gradient through
    # the tangent angle between consecutive pairs of points along the
    # curve. The second option is 'ANALYTICAL' which uses the equation
    # of the curve to calculate the true mathematical gradient from
    # the curve's partial dervivatives

    GRAD_LOSS_TYPE = 'NUMERICAL' # NUMERICAL or ANALYTICAL
```

### DATA_SAMPLING_SCHEME

```python
    # ====================== DATA_SAMPLING_SCHEME ====================== #
    # There are two data sampling schemes. The 'EQUAL' data sampling scheme
    # ensures that in each batch, we have an equal representation of samples
    # from each specific dataset. This schemes over-fits the network on 
    # smaller and underepresented datasets. The second sampling scheme is
    # 'CONCATENATE', in which the data is sampled randomly and the network
    # only sees each image from each dataset once in an epoch

    DATA_SAMPLING_SCHEME = 'EQUAL' # EQUAL or CONCATENATE
```

### BATCH_SIZE_SCHEME

```python
    # ======================== BATCH_SIZE_SCHEME ======================== #
    # There are three type of BATCH_SIZE_SCHEME, the 'CONSTANT' batch size
    # scheme sets a constant, fixed batch size value of 24 throughout training.
    # The 'SLOW_DECAY' batch size scheme reduces the batch size during training,
    # helping the model escape from local minima. The 'FAST_DECAY' batch size
    # scheme decays the batch size faster, and may help with quicker model
    # convergence.

    BATCH_SIZE_SCHEME = 'FAST_DECAY' # FAST_DECAY or SLOW_DECAY or CONSTANT
```
-------------------------------------------

## Visualization:  Loss, Validation Metrics and Network Predictions during training:

If you followed the instructions above, a sub-folder will have been created in the Model/training/ called 'runs'. If you ran the train script from the command line while you were in a different folder, then the 'runs' folder will have been created there. 

Open a separate terminal window and navigate inside the 'runs' folder. You should see a log run with the date, time, and hardware description associated with the training run. To visualize this stored data, execute the tensorboard command and pass in the name of the run to the --logdir parameter as follows:

```bash
tensorboard --logdir=DATE_TIME_YOUR_DEVICE
```

You should see the following printout on the command line:

```bash
Serving TensorBoard on localhost; to expose to the network, use a proxy or pass --bind_all
TensorBoard 2.18.0 at http://localhost:6006/ (Press CTRL+C to quit)
```

Then, open your web browser (e.g. Chrome, Firefox, Edge, etc.) and type the http web address were tensorboard is serving the data, in the case above, it is **http://localhost:6006/** - you should then be able to see a dashboard with the losses being logged, and after a while (once the first 1000 images have been processed), you will also be able to see image visualizations from the network too, it should look something like this:

[image]

-------------------------------------------

## Conducting your training experiment:

Add a comment below, specifying in detail, the state of the modifiable parameters you have set for your training experiment. Once your experiment is finished, please share the best validation score metrics you achieved, including the overall and dataset specific validation metrics, and at which epoch and step you achieved those metrics. Good luck!

TranHuuNhatHuy, 2025-04-29:
## First training attempt - 2025/04/29

### 1. Configs

```
ENDPOINT_LOSS_SCALE_FACTOR = 0.0          
MIDPOINT_LOSS_SCALE_FACTOR = 0.0
GRADIENT_LOSS_SCALE_FACTOR = 1.0
CONTROLPOINT_LOSS_SCALE_FACTOR = 1.0

GRAD_LOSS_TYPE = 'NUMERICAL'
DATA_SAMPLING_SCHEME = 'EQUAL'
BATCH_SIZE_SCHEME = 'FAST_DECAY'
```

- Used GPU: NVIDIA RTX 4070 Ti SUPER

### 2. Status

- Started at: 2025/04/29 1:05 PM JST.
- Abruptedly ended due to runtime error at: 2025/04/29 10:43 PM JST.
- Total step reached: 416,780
- Last epoch: 3

### 3. Best performance

- Dataset validation:

| Dataset | Validation loss |
| --- | --- |
| BDD100K | 0.11341845013559643 |
| COMM2K19 | 0.05767396414293009 |
| CULANE | 0.06726813919671158 |
| CURVELANES | 0.09599979495346118 |
| ROADWORK | 0.07522946281288458 |
| TUSIMPLE | 0.062401770480209964 |
| All | 0.0800096666821073 |

m-zain-khawaja, 2025-04-29:
Thanks @TranHuuNhatHuy - it's great to see the first batch of results.

I have pushed this code-fix (https://github.com/autowarefoundation/autoware.privately-owned-vehicles/commit/523a7df730bca2b339d4060da38aeb3333d754f6) which should resolve the runtime issue.

TranHuuNhatHuy, 2025-05-03:
# Individual Training - TuSimple

### 1. Configs

Same as default.

- Used GPU: NVIDIA RTX 4070 Ti SUPER

### 2. Status

- Manually shut down after ~250k steps in 7.1 hours

### 3. Performance

[image]

TranHuuNhatHuy, 2025-05-04:
# Individual Training - CULane

### 1. Configs

Same as default.

- Used GPU: NVIDIA RTX 4070 Ti SUPER

### 2. Status

- Manually shut down after ~360k steps in 7.6 hours

### 3. Performance

Seems to peak at 200k.

[image]

[image]

[image]

closed 2025-06-05 by m-zain-khawaja